betaflight/bidircfg #1: FC frozen after add  timer b00 1 timer b01 1

RadomirJ, 2019-03-10:


RadomirJ, 2019-03-10:
Motors 1 and 2 doesn't work with dshot bidir, after fix timer settings fc completely frozen.

mikeller, 2019-03-11:
@RadomirJ: Please provide the information that you were asked to provide when opening an issue. Otherwise nobody is going to be able to help you.

RadomirJ, 2019-03-13:
I tried run snippet but motors 1 and 2 still does not spin.

cudakeer808, 2019-03-17:
@RadomirJ , copy the four lines for the timers first then paste it on CLI. Press enter. Do not save YET. Then copy the four lines for dma, paste it in CLI then press enter. Only then you can paste the rest of the snippet.

copy and paste timers. enter. copy and paste dmas. enter. only then paste everything else. enter. save.

SebiWolze, 2019-03-20:
Set up BF 4.0 from scratch
set beeper_frequency = 0
paste snippet into CLI

closed 2019-06-05 by stale[bot]