betaflight/bidircfg #4: BF 4.0 RC2 - DALRC F405 motors 1 and 2 not spinning (Dshot Bidir)

digex, 2019-04-04:
**Describe the bug**
After installing BF4.0  RC2 and blheli FW 32.6.7 , with bidirectional dshot and rpm filters enabled, motor 1 and motor 2 doesn`t spin.

When I disable dshot_bidir all motors spin correctly.

**To Reproduce**
install BF4.0 RC2 and blheli FW 32.6.7 , enable bidirectional dshot and rpm filters.

**Expected behavior**
After arming motor 1 and 2 does`t spin.

This is installed 
[https://github.com/betaflight/bidircfg/blob/master/DALRCF4.cf](url)

```
timer C06 0
timer C07 1

dma pin A10 1
dma pin A08 2
dma pin C08 1

set scheduler_optimize_rate=on
set dshot_burst=off
set dshot_bidir=on

set dyn_notch_width_percent = 0
set dyn_notch_range = medium
set dyn_notch_q = 250

set gyro_lowpass_hz = 150
set dyn_lpf_gyro_min_hz = 150
set dyn_lpf_gyro_max_hz = 575

set dterm_lowpass2_hz = 0

set d_min_pitch=29
set d_min_roll=27
set d_min_boost_gain=35

set debug_mode=rpm_filter

set motor_pwm_protocol=DSHOT600
set gyro_sync_denom = 2
set pid_process_denom = 1

save
```

**status**
```
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.28V, Core temp=50degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3032, Max available config: 16384
GYRO=MPU6000, ACC=MPU6000
System Uptime: 16 seconds, Current Time: 0000-01-01T00:00:00.000+00:00
CPU:14%, cycle time: 254, GYRO rate: 3937, RX rate: 33, System rate: 10
Voltage: 1556 * 0.01V (4S battery - OK)
I2C Errors: 0
Arming disable flags: RXLOSS CLI
Dshot reads: 119976
Dshot invalid pkts: 369
Dshot irq micros: 2
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 4095
Dshot RPM Motor 2: 88
Dshot RPM Motor 3: 78
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
```

**# diff**

```
# version
# Betaflight / DALRCF405 (DLF4) 4.0.0 Mar 14 2019 / 11:09:55 (360afd96d) MSP API: 1.41

# start the command batch
batch start

board_name DALRCF405
manufacturer_id 

# name

# resources

# timer
timer C06 0
timer C07 1

# dma
dma pin C06 NONE
dma pin A10 1
# pin A10: DMA2 Stream 6 Channel 6
dma pin A08 2
# pin A08: DMA2 Stream 3 Channel 6
dma pin C08 1
# pin C08: DMA2 Stream 4 Channel 7
dma pin C07 NONE

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_max_hz = 575
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON

# profile
profile 0

set dterm_lowpass2_hz = 0
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0

# end the command batch
batch end
```

VirtualEnder, 2019-04-04:
Fixed in PR #3 waiting on merge.

bobbfwed, 2019-04-09:
Just tested it with RC6 and the newest version of commands on https://github.com/betaflight/bidircfg/blob/master/DALRCF4.cf and it worked great!

Rclf, 2019-04-12:
same here on clracingf4 ,motors 1 and 2 do not spin

Dshot reads: 6938
Dshot invalid pkts: 0
Dshot irq micros: 3
Dshot RPM Motor 0: 4095
Dshot RPM Motor 1: 4095
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0

closed 2019-06-05 by stale[bot]