betaflight/bidircfg #6: RPM filter motor grinding

robert-hoffmann, 2019-04-04:
With the laters build of BF (betaflight_4.0.0_MATEKF405-1509) and BLHELI32 (Furling32_Multi_32_67), on StingerSwarm 2207 2150KV motors.

I get slight grinding noise on the motors after following the setup guide (to the letter)
https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter

Using 4k/4k

..i did revert gyro_lowpass_hz to the default value of 150, but nothing changed.

This does not happen when not using the RPM filter, and even more wierd, i get no grinding noise when spinning the motors via the BLHELI interface, only from within the configurator.

I also lowered D a little, but that did not help.

Also, once when i spun the motors in BLHELIsuite, and then went straight to BF configurator, the grinding dissapeared, then i unplugged everything (usb+batt ..i do this systematically after each change) and it came back.

It's not horribly bad ..but any idea if this could simply be a setting that could be tweaked, or something else ?

Thanks for any feedback !

**UPDATE**
Went from a config that works without any grinding sound to the RPM filter mode by **only** changing the following lines (no messing with filters) and the grinding still appeared ..so it's not related to turning various filters off:
```
set dshot_burst=off
set scheduler_optimize_rate=on
set dshot_bidir=on
set gyro_sync_denom = 2
```

---
# version
# Betaflight / MATEKF405 (MKF4) 4.0.0 Mar 11 2019 / 11:58:49 (0deca9c6f) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405
manufacturer_id 
mcu_id 001b00323436510639333434
signature 

# name
name Chameleon

# resources
resource MOTOR 5 NONE
resource SERIAL_TX 11 A15

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 4 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1950 2050 0 0
aux 1 13 0 1950 2050 0 0
aux 2 35 2 1950 2050 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 150
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 0
set acc_hardware = NONE
set acc_calibration = 18,30,-23
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 8
set airmode_start_throttle_percent = 20
set blackbox_device = NONE
set dshot_idle_value = 400
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set ibata_scale = 167
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set yaw_deadband = 1
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_rssi_pos = 2138
set osd_tim_2_pos = 2081
set osd_current_pos = 2496
set osd_mah_drawn_pos = 2520
set osd_warnings_pos = 2505
set osd_avg_cell_voltage_pos = 2104
set osd_stat_max_fft = ON
set scheduler_optimize_rate = ON
set vtx_band = 3
set vtx_channel = 8
set vtx_power = 3
set vtx_freq = 5945
set vcd_video_system = PAL

# profile
profile 0

set dyn_lpf_dterm_max_hz = 300
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 4000
set feedforward_transition = 45
set iterm_rotation = ON
set iterm_relax_type = GYRO
set d_pitch = 33
set f_pitch = 65
set d_roll = 30
set f_roll = 60

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 30
set thr_expo = 60
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 20
set roll_srate = 85
set pitch_srate = 85
set tpa_rate = 80
set tpa_breakpoint = 1750

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
---

robert-hoffmann, 2019-04-14:
Just an update

Tried again with BF 4.1 build 1569, and blheli Furling32_Multi_32_61.Hex

Issue is still there, and as you can hear in the video above, you can see jerkiness in the gyro tab in BF when the noise appears.

robert-hoffmann, 2019-04-14:
Did some further testing after checking the latest blheli docs

I'm pretty sure i did this in a previous version ..but just now it worked

**Turn "sine modulation mode" to OFF in blheli**

This seems to get rid of the grinding noises, so i will test this in the field as soon as i can, but probably before next week or so, as i'm waiting for a replacement motor, and it's gonna be really windy this week anyways..

robert-hoffmann, 2019-04-15:
Not sure if the problem is on BF or blheli side, but sine mode must be interfering in some way with packets sent via bidir.

Like i said, i tested this all the way down to 24khz pwm, and as far as esc capabilities go (72Mhz & FPU F3), i suppose resource-wise, this should pass.

robert-hoffmann, 2019-05-04:
I confirm this grinding issue is when sine mode modulation is activated at the same time as dshot bdir and running the rpm filters.

Fix: turn off sine mode modulation in blheli

closed 2019-05-04 by robert-hoffmann

robert-hoffmann, 2019-06-04:
They fixed this in 32.6.4 (normal dshot) and 32.6.5 (inverted dshot)

yoodog, 2019-10-04:
Great read, thanks for follow up! Was researching whether to activate Sine Modulation or not - in theory no drawbacks for Freestyle as far as I understand. What are your thoughts after all those tests?

yoodog, 2019-10-04:
> They fixed this in 32.6.4 (normal dshot) and 32.6.5 (inverted dshot)