betaflight/bidircfg #11: Need to set resource beeper none for rpm filter to work on CLRACINGF4

mmakam2, 2019-04-14:
The rpm filter wiki didn't mention this but you need to set resource for the beeper to none for the motors to all spin. Not sure if this is the Bardwell board or all CLRACINGF4 boards. 

Requesting to get this added to the wiki.

mikeller, 2019-04-14:
@mmakam2: The wiki is editable by all GitHub users - you can add this information to the wiki yourself.

ShineT0T, 2019-04-22:
when i put in the settings to enable the rpm filter my beeper just does a continuos beep when bat is connected, regardless if resource beeper 1 none or b04 (default setting). might my controller still use b04 for the beeper even when unmapped? beeper shouldnt work at all when its resource is set to none, should it? 
it seems like some of the b00 etc are mapped to more than one  funktion, is that correct? 
this is my first time mapping so id rather ask than try to fly it on

ShineT0T, 2019-04-22:
soo after long reading and searching i did find that you actually do not need to set the resource beeper 1 none command. rather do set beeper_frequency=0 before pasting the snippet. beeper and led strip should work fine as far as i understood.

closed 2019-06-05 by stale[bot]