betaflight/bidircfg #16: Issue with bidirectional dshot (motors unresponsive)

drdrechronik, 2019-04-15:
Seems to be an issue with Bidirecitonal dshot. When active (set dshot_bidir = ON), i dont't get the "valid throttle signal beeps". Thus I cant spin the motors, neither using the betaflight configurator, nor using blheli suite. 
also. when typing the "status" command in the cli, it seems that the fc locks up, I don't get any response in the cli and i have to power-cylce the fc.
Alos when using the "tasks" command, it shows 3999Hz for loop time.
ESCs (Airbot_Wraith32_ST_Multi_32_61) are on the latest test firmware of BLHELI. PWM is on 48Khz, Auto timing is enabled and LED (on ESCs) are set to a color.

FYI the default dshot rate I run is dshot600, but I set it to dshot150 to see if it resolves things, unfortunately, the problem persists regardless which dshot i select.

Would love to try the rpm filter though... Any suggestions?

# diff

# version
# Betaflight / MATEKF722SE (MF7S) 4.0.0 Apr 11 2019 / 07:26:45 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id 

# name

# resources
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 B05
resource MOTOR 4 B04
resource CAMERA_CONTROL 1 A01

# timer

# dma
dma pin B04 NONE
dma pin B05 NONE
dma pin B00 NONE
dma pin B01 NONE
dma pin A15 NONE
dma pin B03 NONE
dma pin B06 NONE
dma pin B07 NONE
dma pin A08 NONE
dma pin A01 NONE
dma pin A00 NONE

# mixer

# servo

# servo mix

# feature
feature GPS
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 2 115200 9600 0 115200
serial 2 8192 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1900 2100 0 0
aux 1 46 1 900 1100 0 0
aux 2 13 1 1900 2100 0 0
aux 3 35 1 1700 1900 0 0
aux 4 39 1 1550 1700 0 0
aux 5 40 1 1300 1450 0 0
aux 6 49 1 1100 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 0
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 150
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set acc_hardware = NONE
set acc_calibration = -36,122,309
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_derivative_hz = 150
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_idle_value = 350
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT150
set align_board_yaw = 180
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 30
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 1
set osd_warn_no_gps_rescue = OFF
set osd_cap_alarm = 1000
set osd_alt_alarm = 50
set osd_vbat_pos = 2487
set osd_rssi_pos = 2477
set osd_link_quality_pos = 430
set osd_tim_1_pos = 2508
set osd_tim_2_pos = 33
set osd_flymode_pos = 35
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2466
set osd_mah_drawn_pos = 2498
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 2102
set osd_home_dir_pos = 2092
set osd_home_dist_pos = 2095
set osd_compass_bar_pos = 266
set osd_altitude_pos = 2083
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2519
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_max_fft = ON
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_channel = 4
set vtx_freq = 5800

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 7000
set feedforward_transition = 10
set p_pitch = 55
set d_pitch = 30
set p_roll = 45
set d_roll = 30
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0

set roll_rc_rate = 108
set pitch_rc_rate = 108
set yaw_rc_rate = 120
set roll_expo = 12
set pitch_expo = 12
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 75

# end the command batch
batch end

drdrechronik, 2019-04-15:
no, now with the timers set correctly, when I disable dshot_bidir i cant arm. nor does blheli correctly show my escs correctly. when loading my escs in blheli it shows dashes instead of Wraith blahblahblah...

dadoribeiro, 2019-04-16:
When something like this happened to me (same ESC) I reinstalled the latest release BF 4.0 with full erase and left all the defaults except motor protocol to DSHOT 600. Then I tried connecting to BLHeli32 and flashed to the stable firmware making sure all 4 were flashed correctly and left then at all defaults, just changing motor directions as needed. If it arms then I started setting up bdir: first flashed ESCs one by one to the test firmware and using defaults again. Then you test arm it again. If it does then you put the snippet, but only the lines for timer, dma and the 3 beneath the “basic settings” and saving. Then you plug the lipo and hear the beeps, if it is ok you connect to usb and pull the cli status. I was able to save my ESCs this way

drdrechronik, 2019-04-22:
I followed your advice and the rpm filter works fine. I suspect that it had something to do with the dma settings (as seen in the top section of the diff) not all of these were part of the snippet.

Thanks

closed 2019-04-22 by drdrechronik