betaflight/bidircfg #17: CLRacing F7 motors 3 & 4 not spinning with RPM filtering

serran0, 2019-04-17:
While running BF4.0 release version, and all previous RC's - the motors 3 and 4 dont spin.
Using the default snippet on a clean flash.

set gyro_sync_denom = 2
set pid_process_denom = 1
set motor_pwm_protocol=DSHOT600
set scheduler_optimize_rate=on
set dshot_burst=off
set dshot_bidir=on

Motors 1 and 2 will spin, but up to a certain throttle level, then the motor stops and no motors at all will spin until reboot.

# status
MCU F722 Clock=216MHz, Vref=3.30V, Core temp=69degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3028, Max available config: 16384
GYRO=ICM20602
System Uptime: 12 seconds, Current Time: 2019-04-17T21:42:00.861+00:00
CPU:3%, cycle time: 250, GYRO rate: 4000, RX rate: 33, System rate: 10
Voltage: 2359 * 0.01V (6S battery - OK)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP RPMFILTER
Dshot reads: 7617
Dshot invalid pkts: 2538
Dshot irq micros: 3
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
7 7 7 7 7 14 7 7 7 7 14 14 7 14 14 14 0 0 14 1 15 2 16 3 17 4 17 5 18 6 19 6 
0 0 7 0 0 0 7 0 0 7 7 7 7 8 8 7 

# diff

# version
# Betaflight / CLRACINGF7 (CLR7) 4.0.0 Apr 11 2019 / 07:02:11 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name CLRACINGF7
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set acc_hardware = NONE
set mag_hardware = NONE
set serialrx_provider = FPORT
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set yaw_motors_reversed = ON
set pid_process_denom = 1
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON

# profile
profile 0

set dterm_lowpass2_hz = 0

# rateprofile
rateprofile 0

# end the command batch
batch end

Also tested enabling bi-dir DSHOT with current dev fw 4.1.0 #1574 but the board will then not boot at all.

Target: CLRACINGF7
Name/Version: 4.1.0 #1574
Binary: betaflight_4.1.0_CLRACINGF7-1574-55a5d70cc.hex
Date: 16-04-2019 20:42

danielschafer, 2019-04-23:
I have made exactly the same observation.

danielschafer, 2019-04-25:
I finally figured it out.
I had to reflash the ESCs, since two of them weren't flashed properly.
And I had to remap MOTOR 4 to the LEDS Pad.
Now it works like intended.

serran0, 2019-04-26:
Yep, remapped Motor 4 to led_strip (B01) and now both 3& 4 motors are working with bidir.

closed 2019-04-26 by serran0

joelucid, 2019-09-21:
I added the following commands to the CLRACINGF7 snippet:

```
resource led 1 none
resource motor 4 b01
```