betaflight/bidircfg #26: Talon F7 V2 no bidirectional DSHOT possible

MartinFPV, 2019-05-20:
**Describe the bug**
When trying to activate the bidirectional DSHOT under Betaflight. Turn on 2 motors only briefly, then nothing happens anymore. Before activating the Bidirectional DSHOT, all 4 engines were unremarkable.

**To Reproduce**
https://youtu.be/rpyD-B7AGoQ

**Expected behavior**
All 4 motors should behave normally with activated bidirectional DSHOT.

**Flight controller configuration**
``` 
# diff
# version
# Betaflight / CLRACINGF7 (CLR7) 4.0.2 May 5 2019 / 11:41:04 (56bdc8d26) MSP API: 1.41
# start the command batch
batch start
board_name CLRACINGF7
manufacturer_id
# name
# resources
# timer
# dma
# mixer
# servo
# servo mix

# feature
# beeper
# beacon
# map
# serial
# led
# color
# mode_color
# aux
# adjrange
# rxrange
# vtx
# rxfail
# display_name
# master
set gyro_sync_denom = 2
set acc_hardware = NONE
set mag_hardware = NONE
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set vbat_scale = 81
set pid_process_denom = 1
set scheduler_optimize_rate = ON
# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end
#
```
**Setup / Versions**
 Talon F7 BB V2 20x20  @BF 4.0.2 (Target CLRACINGF7)
 Furling32 4in1 mini ESC - F3 32bit BLHELI ESC  @ BLHeli_32 Test code Rev32.6.2 (Target Furling32_4in1_Multi_32_62.Hex)
[image]

 

**Additional context**
Add any other context about the problem here.

MartinFPV, 2019-05-21:
But if that were the case, the same hardware conditions apply for the TALON F7 as for the CLRACINGF7, why do only 2 motors run for a short time. Then 3 motors would have to start shortly, right? I think that there is more behind.

To the Schreenschots:
bottom with bidirectional dshot enabled.
above without activated bidirectional dshot

[image]
[image]

mikeller, 2019-05-21:
@MartinFPV: Yeah, looks like this is an incorrect clone of the CLRACINGF7. Since the manufacturer does not provide their own target, or collaborate with us on developing one, there is not much we can do to fix this. You should probably take this up with the manufacturer.

MartinFPV, 2019-05-22:
Okay, thanks for the information, I wrote HeliNation's support today.

closed 2019-06-28 by stale[bot]