betaflight/bidircfg #32: problems with mamba f722 mini (packet loss and no current sensing)

m013, 2019-07-14:
tried 4.0.4 and 4.1 #1746
blheli versions Rev32.6.4 and Rev32.6.5 respectively

the first problem - huge packet loss (up to 40%) i cant even arm (rpmfilter warning).
with Rev32.6.5 two of the motors get no data

the second - although i have telemetry readings from the esc on osd (rpm and temperature), i have no current sensing. i know you have to apply some load on motors tjo get readings, but i did it via motor test tab and still nothing. note that there is no current calibration setting in blheli suite
[dump.txt](https://github.com/betaflight/betaflight/files/3389729/dump.txt)
[image]
[image]

jorgemia, 2020-12-28:
@m013 were you able to get the current sensor working? I've set the current to Onboard ADC (as I'm using the current cable from ESC) but getting 0V when plugging the battery.

m013, 2020-12-28:
> 
> 
> @m013 were you able to get the current sensor working? I've set the current to Onboard ADC (as I'm using the current cable from ESC) but getting 0V when plugging the battery.
have you tried to fly ?maybe it doesnt output small currents

jorgemia, 2020-12-28:
@m013 But I'm not even getting a voltage reading, everything is at 0.

m013, 2020-12-28:
@jorgemia  i had no problems with voltage back then