betaflight/bidircfg #33: Mamba F722 Mini RMP Filtering Not Working ... Only 2 Motors Spin Up

TheFrisco, 2019-07-14:
BUG DESCRIPTION
After following the instructions and using the default snippet, only motors #1 and #2 spin up. Also, the invalid pkts are much higher than 1%. Using the Mamba F722 Mini stack, both ESC and FC, installed the correct BLHeli 32 firmware per instructions so a little baffled. And YES the table doesn't include the Mamba stack but was hoping you can help anyhow. THANKS!

HOW TO REPRODUCE
Use Mamba F722 Mini stack, follow instructions and try to spin up motors in BF. Only #1 and #2 spin up.

BF DIFF
'''
# version
# Betaflight / MAMBAF722 (MBF7) 4.0.4 Jun 30 2019 / 14:33:38 (f3a95efa3) MSP API: 1.41

# start the command batch
batch start

board_name MAMBAF722
manufacturer_id 

# name
name TheFrisco

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 0 115200 57600 0 115200
serial 4 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1700 2100 0 0
aux 1 1 2 900 1700 0 0
aux 2 13 2 900 1300 0 0
aux 3 35 2 900 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 0
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set rssi_channel = 8
set serialrx_provider = CRSF
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 9
set bat_capacity = 550
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_cap_alarm = 550
set osd_vbat_pos = 2433
set osd_rssi_pos = 2106
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2349
set osd_vtx_channel_pos = 33
set osd_crosshairs_pos = 2253
set osd_craft_name_pos = 2090
set osd_gps_speed_pos = 394
set osd_gps_sats_pos = 89
set osd_home_dir_pos = 398
set osd_home_dist_pos = 399
set osd_avg_cell_voltage_pos = 2401
set osd_stat_max_dist = ON
set osd_stat_battery = ON
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vcd_video_system = NTSC

# profile
profile 0

set dterm_lowpass2_hz = 0
set f_pitch = 120
set f_roll = 120
set f_yaw = 50
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0

set thr_expo = 40
set roll_rc_rate = 90
set pitch_rc_rate = 90
set yaw_rc_rate = 90
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 76
set pitch_srate = 75
set tpa_rate = 75

# end the command batch
batch end

'''
SETUP/VERSIONS
 - Mamba F722 stack including 32bit ESC, from RaceDayQuad

m013, 2019-07-15:
https://github.com/betaflight/bidircfg/issues/32

Unsstable, 2019-07-16:
Diatone bring the configuration that works, last Sunday I contact with the leader of the company and he solve the problem. Search in the Facebook page and you will find it.

https://m.facebook.com/groups/529347843890272?view=permalink&id=1355920687899646

Hope it works for you too!

m013, 2019-07-16:
@Unsstable  it is a closed group

TheFrisco, 2019-07-16:
> Diatone bring the configuration that works, last Sunday I contact with the leader of the company and he solve the problem. Search in the Facebook page and you will find it.
> 
> https://m.facebook.com/groups/529347843890272?view=permalink&id=1355920687899646
> 
> Hope it works for you too!

Can you please post the resolution here, the link above doesn't work. THANKS!!!

m013, 2019-07-16:
@TheFrisco 
target specific part -

dma pin a09 1
dma pin a08 1 
dma pin c09 0
dma pin c08 1

it works for the mini and fullsized

TheFrisco, 2019-07-16:
> @TheFrisco
> target specific part -
> 
> dma pin a09 1
> dma pin a08 1
> dma pin c09 0
> dma pin c08 1
> 
> it works for the mini and fullsized

So no re-soldering needed, just put this into CLI?

Thanks Robert

m013, 2019-07-16:
no soldering

TheFrisco, 2019-07-16:
PERFECT ... All motors spinning now. Gonna go out today and fly it for real testing too. Then we should add this to the table in the RPM filtering list.

Thanks a lot @m013

TheFrisco, 2019-07-17:
SOLUTION ... TESTED & PERFECT
Goto CLI and paste this:

dma pin a09 1
dma pin a08 1
dma pin c09 0
dma pin c08 1
save

Thanks for the help, especially @m013 !!!

closed 2019-07-17 by TheFrisco

TheFrisco, 2019-07-17:
@etracer65 tried to update the Wiki page but think I screwed it up. In any case, I think the Mamba F722 works now, with the above CLI additions to the DEFAULT.cf

Thanks for all you do!

SKNteknix, 2019-07-23:
Hi guys. 
Im new in this forum but anyway i got the same problem as above with suddenly only motor 1 and 2 is spinning up when arming. Tried to paste in 

dma pin a09 1
dma pin a08 1
dma pin c09 0
dma pin c08 1

into CLI but just get the responce 

Entering CLI Mode, type 'exit' to return, or 'help'

# dma pin a09 1
###ERROR: INVALID DMA OPTION FOR PIN A09: '1'###

# dma pin a08 1
###ERROR: INVALID DMA OPTION FOR PIN A08: '1'###

# dma pin c09 0
# dma C09: no change: 0

# dma pin c08 1
###ERROR: INVALID DMA OPTION FOR PIN C08: '1'###

Any ideas ..? 

Thx

TheFrisco, 2019-07-23:
Sorry to hear @SKNteknix totally worked for me. You sure you have the Mamba 32bit, F7 stack with the MAMBAF722 firmware?

If not, try joining the FB Diatone forum and post there. Here is the link ... https://www.facebook.com/groups/diatoneinnovations/

Good luck and don't give up, totally worth it!