betaflight/bidircfg #40: Matek F411: BF 4.1 DShot BitBang not working when assigning Pin A08 (LED) to cam control

RipperDrone, 2019-10-14:
BetaFPV Whoop AIO 2-4S with Matek F411 FC, BF 4.1: 
BF 4.1 DShot BitBang not working when assigning Pin A08 (LED) to cam control

Steps to reproduce the behavior:
- flash BF 4.1 Nightly (today's first)
- flash JESC, rpm telemetry (v2.1, 24kHz version) to ESCs
- paste universal snippet to enable rpm telemetry and DSHOT300 etc. (from BF 4.1)
- CLI this:
# resource LED_STRIP 1 NONE
# resource CAMERA_CONTROL 1 A08

Everything worked ok until I remapped the LED_STRIP pin. Seems timers are used differently now, because it did work in non bit bang mode before.

joelucid, 2019-10-16:
@RipperDrone could you verify whether this is fixed now that #8978 has been merged?

joelucid, 2019-10-16:
Perfect, thanks for testing!