betaflight/bidircfg #41: Role of death with Bidirectional DSHOT on H7EXTREME

maGfpv, 2019-10-14:
**Describe the bug**
Role of death with activated Bidirectional DSHOT and RPM Filter on SPRACINGH7EXTREME

**To Reproduce**
Bidirectional DSHOT was fixed with this issue betaflight/betaflight#8836. 
I updated to Build betaflight/betaflight#1900 (28.09.2019 22:50:24) and activated rpm filter. Everything else is set to default. Sometimes it happens that the quad performs a role of death. (Log is attached [LOG00008.zip](https://github.com/betaflight/betaflight/files/3726415/LOG00008.zip))

**Flight controller configuration**
```# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Sep 28 2019 / 23:25:54 (fcaa16cd7) MSP API: 1.42

# start the command batch
batch start

board_name SPRACINGH7EXTREME

# name: Marmotte

# resources
resource MOTOR 1 A03
resource MOTOR 2 A02
resource MOTOR 3 A01
resource MOTOR 4 A00

# feature
feature -TRANSPONDER

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# serial
serial 0 64 115200 57600 0 115200
serial 1 32 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# led
led 0 11,11::C:2
led 1 12,12::C:2
led 2 13,13::C:2
led 3 14,14::C:2
led 4 15,15::C:2
led 5 5,15::C:2
led 6 6,15::C:2
led 7 7,15::C:2
led 8 8,15::C:2
led 9 9,15::C:2
led 10 4,11::C:2
led 11 3,12::C:2
led 12 2,13::C:2
led 13 1,14::C:2
led 14 0,15::C:2

# aux
aux 0 0 0 900 1300 0 0
aux 1 13 2 1700 2100 0 0
aux 2 15 6 900 1075 0 0
aux 3 35 4 900 1300 0 0

# adjrange
adjrange 1 0 6 1100 2100 30 6 0 0

# rxrange
rxrange 0 987 2011
rxrange 1 990 2011
rxrange 2 987 2011
rxrange 3 990 2011

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 13 20 26
vtxtable powerlabels 25 100 400

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set mag_hardware = NONE
set min_check = 1010
set max_check = 1990
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 180
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 360
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 1
set ledstrip_race_color = GREEN
set osd_vbat_pos = 384
set osd_rssi_pos = 2106
set osd_tim_2_pos = 2423
set osd_current_pos = 406
set osd_mah_drawn_pos = 2456
set osd_altitude_pos = 2401
set osd_warnings_pos = 14730
set osd_avg_cell_voltage_pos = 2433
set osd_disarmed_pos = 2347
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_max_fft = ON
set debug_mode = RPM_FILTER
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5917
set vcd_video_system = NTSC
set name = Marmotte

profile 0

# profile 0
set vbat_pid_gain = ON
set feedforward_transition = 30
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 10

rateprofile 0

# rateprofile 0
set roll_rc_rate = 95
set pitch_rc_rate = 95
set yaw_rc_rate = 105
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 20
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82

# end the command batch
batch end
```

```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: MOTOR 4
A01: MOTOR 3
A02: MOTOR 2
A03: MOTOR 1
A04: FREE
A05: FREE
A06: VTX_DATA 10
A07: VTX_CLK 10
A08: LED_STRIP
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: GYRO_CS 1
B00: VTX_CS 10
B01: VTX_POWER
B02: QSPI_CLK 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: FREE
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: QSPI_BK1CS 1
B11: FREE
B12: GYRO_CS 2
B13: SERIAL_TX 5
B14: FREE
B15: SERIAL_RX 1
C00: ADC_CURR
C01: ADC_BATT
C02: SPI_MISO 2
C03: SPI_MOSI 2
C04: ADC_RSSI
C05: ADC_EXT
C06: FREE
C07: FREE
C08: SDIO_D0
C09: SDIO_D1
C10: SDIO_D2
C11: SDIO_D3
C12: SDIO_CK
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: SDIO_CMD
D03: SPI_SCK 2
D04: GYRO_EXTI
D05: SERIAL_TX 2
D06: SPI_MOSI 3
D07: BEEPER
D08: FREE
D09: FREE
D10: SDCARD_DETECT
D11: QSPI_BK1IO0 1
D12: QSPI_BK1IO1 1
D13: QSPI_BK1IO3 1
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: QSPI_BK1IO2 1
E03: LED 1
E04: SYSTEM
E05: CAMERA_CONTROL
E06: FREE
E07: QSPI_BK2IO0 1
E08: QSPI_BK2IO1 1
E09: QSPI_BK2IO2 1
E10: QSPI_BK2IO3 1
E11: OSD_CS
E12: SPI_SCK 4
E13: SPI_MISO 4
E14: SPI_MOSI 4
E15: GYRO_EXTI
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE
G00: FREE
G01: FREE
G02: FREE
G03: FREE
G04: FREE
G05: FREE
G06: FREE
G07: FREE
G08: FREE
G09: FREE
G10: FREE
G11: FREE
G12: FREE
G13: FREE
G14: FREE
G15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: MOTOR 4
    CH2: MOTOR 3
    CH3: MOTOR 2
    CH4: MOTOR 1
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE
TIM15:
    CH1: CAMERA_CONTROL
TIM16: FREE
TIM17: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 4
DMA1 Stream 1: MOTOR 3
DMA1 Stream 2: MOTOR 2
DMA1 Stream 3: MOTOR 1
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: ADC 3
DMA2 Stream 2: LED_STRIP
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

 
```
# dshot_telemetry_info
Dshot reads: 268399
Dshot invalid pkts: 466
Dshot directionChange cycles: 984, micros: 2

Motor      eRPM      RPM      Hz   Invalid
=====   =======   ======   =====   =======
    0         0        0       0     0.66%
    1         0        0       0     0.00%
    2         0        0       0     0.00%
    3         0        0       0     0.00%

117 133 149 165 197 213 229 246 278 294 310 326 342 358 406 14 0 0 3373138432 1990328054 572784642 4290238063 
16 16 16 32 16 16 17 32 16 16 16 16 16 48 4294966904 4294967282 0 3373138432 2912156918 2877423884 3717453421 

# 
```

```
tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9      12       0    0.5%    0.0%         0
01 - (         SYSTEM)    999      14       1    1.8%    0.5%       124
02 - (       GYRO/PID)   8002      55      47   44.5%   38.1%     41653
03 - (            ACC)    999      26       9    3.0%    1.3%      1057
04 - (       ATTITUDE)     99      18       4    0.6%    0.5%        58
05 - (             RX)     32      34      20    0.6%    0.5%        81
06 - (         SERIAL)     99  314493       0 3113.9%    0.0%       665
07 - (       DISPATCH)    998      13       0    1.7%    0.0%        30
08 - (BATTERY_VOLTAGE)     49      14       1    0.5%    0.5%         6
09 - (BATTERY_CURRENT)     49      12       1    0.5%    0.5%         2
10 - ( BATTERY_ALERTS)      4      13       1    0.5%    0.5%         0
11 - (         BEEPER)     99      12       1    0.6%    0.5%         6
14 - (           BARO)    153      30       6    0.9%    0.5%       112
15 - (       ALTITUDE)     39      14       3    0.5%    0.5%        11
17 - (      TELEMETRY)    249      24       0    1.0%    0.0%        81
18 - (       LEDSTRIP)     99      34       3    0.8%    0.5%        40
20 - (            OSD)     59     234      13    1.8%    0.5%       104
22 - (            CMS)     59      11       1    0.5%    0.5%         3
23 - (        VTXCTRL)      4      25       0    0.5%    0.0%         2
24 - (        CAMCTRL)      4      12       0    0.5%    0.0%         0
26 - (    ADCINTERNAL)      1      12       1    0.5%    0.5%         0
27 - (       PINIOBOX)     19      12       0    0.5%    0.0%         1
RX Check Function                   2       0                         2
Total (excluding SERIAL)                        62.3%   45.9%

# 
```

**Setup / Versions**
 - Flight controller: H7EXTREME
 - ESC: TekkoF3 Metall 65A with 32.7.0 of blheli32
 - Esc are connected via extra ground and pwn wires not twisted
-F40 pro III 1600kv

**Additional context**
Here is a short video: https://youtu.be/7YiYsa2Esek

Mateyhv, 2019-10-15:
Adding to previous comment, I haven't flown yet, but I can still reproduce the unsynchronized esc initialization sound which if this hapens then repeats (like an ESC reboot) the beeps but even more unsynchronized, almost one ESC after the other. Noticed doing several enter/exits from CLI to update config after the BF4.1 stable release. This happened to me before the H7 rpm fix, and still happens so there is still something wrong somewhere. @magFPV might have had an ESC reboot midair?

maGfpv, 2019-10-15:
I also noticed the unsynchronized esc initialization beeps when entering the cli. 

And i don't know if it's a esc reboot. Could this be related to bidir/rpm filter? because it never happened before.

nerocide, 2019-10-16:
Hi guys,
Experiencing the same kind of behaviour with **RPMFilter** enabled on **H7 Extreme**.
Using **Tekko32 F3 45A Slim** (separated) with Blheli **32.7**

I've tried the following:

RPM Filter | Gyro Update | PID loop | Gyro to use  | DSHOT | ESC startup tones | Result
------------ | ------------- | ------------- | ------------- | ------------- | -------------| -------------
ON | 8K | 8K | BOTH | 600 | All unsynced, almost starting one after the other | Dead roll on hard moves
ON | 8K | 8K | FIRST | 600 | All unsynced, almost starting one after the other | Dead roll on hard moves
ON | 8K | 8K | BOTH | 300 | sometime unsynced, but most of the time synced | Dead roll on hard moves but less often
ON | 8K | 8K | FIRST | 300 | sometime unsynced, but most of the time synced | No problem yet, test to be continued

I will try so other settings tomorrow probably to be sure of the behavious an reproductivity.

joelucid, 2019-10-16:
I setup my h7extreme and a typhoon32 (32.7) on the bench. Startup tones were synchronized at 8k8k, dshot 600, both gyros with bidirectional dshot on. 

Looking at the log:

[image]

Looks like communication with the escs breaks down exactly at the cursor. We get one rpm value of 3494 hz which is obviously way too high and then at least the back channel breaks down.

Could you fly with debug-mode=DSHOT_RPM_ERRORS so we can see if this might be a signal quality issue?

maGfpv, 2019-10-16:
Sure i will do, hopefully this weekend

nerocide, 2019-10-16:
> Could you fly with debug-mode=DSHOT_RPM_ERRORS so we can see if this might be a signal quality issue?

I'll try ASAP.

Mateyhv, 2019-10-16:
From @nerocide test table I am on the third row although running 4K/4K. Just by bench testing, ESC is Mamba 506 4in1:

ON | 4K | 4K | BOTH | 300 | _roughly 50% unsynced_

mikeller, 2019-10-17:
@Mateyhv, @maGfpv:

> I also noticed the unsynchronized esc initialization beeps when entering the cli.

> And i don't know if it's a esc reboot. Could this be related to bidir/rpm filter? because it never happened before.

This is most likely due to the code for auto-completion in CLI that has been added to configurator, which is probing a number of commands when CLI is entered.

You can verify this by disabling 'Advanced CLI AutoComplete' in the top right 'gear' menu in configurator and checking if this makes it go away.

haslinghuis, 2019-10-17:
Flashed H7 to from 4.1RC6 to 4.1. Diff added. Set debug filter to dshot_rpm_errors. When I used throttle to takeoff I got a roll of death and my propeller broke... again. 

I am planning to upgrade the frame to a normal size anyway, because the props are not cheap and I am not allowed to fly above 250gram anyway. This was my 12" hexacopter build which became a 10" 1kg quad with 6:1 TWR.

[H7_DSHOT_RPM_ERROR.zip](https://github.com/betaflight/bidircfg/files/3739270/H7_DSHOT_RPM_ERROR.zip)

joelucid, 2019-10-17:
Hmm - no errors in that log at all. The log looks kinda funky:

[image]

Lots of motor activity on all motors but Motor 1 and then the pids go a bit crazy. This doesn't look like a death roll.

joelucid, 2019-10-17:
Weird. Here is a larger view:

[image]

haslinghuis, 2019-10-17:
Well soon after I raised the throttle it responded with a hard flip (rolling in left direction) - very quick. I have no props leftovers - I could do 2x8" front and 2x10" rear? I did not change the PID settings from last time nor did I reapply them, only the notches we had figured out last flight.

nerocide, 2019-10-18:
Yesterday evening on 4 packs i wasn't able to reproduce deathrolls.
Probably because i've redone the entire configuration from scratch after **rewriting official FW 4.1**.

Still using **TEKKO32 F3 Slim 45A** on Blheli 32.7
Only thing i added was **set idle_min_rpm = 19** (except my rated, OSD, vtxtable..)

I've noticed that the ESC stills starts with delay sometime, even yesterday. 
**BUT all my flights were done when ESC initialization were synced.**

Also noticed **dshot_telemetry_info** shows 0.5% error on motor 4 every time. And i had it running by pulse on the bench after plugging in a lipo, this happened once only and wasn't able to reproduce since.

Here is my diff:

```terminal
# version
# Betaflight / SPRACINGH7EXTREME (SP7E) 4.1.0 Oct 16 2019 / 11:56:09 (c37a7c91a) MSP API: 1.42

# start the command batch
batch start

board_name SPRACINGH7EXTREME

# name: XbeeLB1_LR_6P

# feature
feature -RSSI_ADC
feature -LED_STRIP
feature GPS

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1300 1700 0 0
aux 4 35 2 1700 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
[... 95 lines cut ...]
03 - (            ACC)    998      23       7    2.7%    1.1%       953
04 - (       ATTITUDE)    499      17       4    1.3%    0.6%       262
05 - (             RX)     32      39      19    0.6%    0.5%        75
06 - (         SERIAL)     99  319999       0 3168.4%    0.0%       658
07 - (       DISPATCH)    998      13       0    1.7%    0.0%        26
08 - (BATTERY_VOLTAGE)     49      14       2    0.5%    0.5%         6
09 - (BATTERY_CURRENT)     49      12       0    0.5%    0.0%         1
10 - ( BATTERY_ALERTS)      4      13       2    0.5%    0.5%         0
11 - (         BEEPER)     99      14       2    0.6%    0.5%        15
12 - (            GPS)     99      51       7    1.0%    0.5%        81
14 - (           BARO)    153      29       6    0.9%    0.5%       106
15 - (       ALTITUDE)     39      15       2    0.5%    0.5%        10
17 - (      TELEMETRY)    499      21       0    1.5%    0.0%       186
19 - (    TRANSPONDER)    249      13       0    0.8%    0.0%         9
20 - (            OSD)     59     267      20    2.0%    0.6%       143
22 - (            CMS)     59       2       0    0.5%    0.0%         2
23 - (        VTXCTRL)      4      24       1    0.5%    0.5%         3
24 - (        CAMCTRL)      4      11       0    0.5%    0.0%         0
26 - (    ADCINTERNAL)      1      12       0    0.5%    0.0%         0
27 - (       PINIOBOX)     19      12       0    0.5%    0.0%         1
RX Check Function                   3       1                         3
Total (excluding SERIAL)                        60.2%   37.1%
```
I took BB for all my fights (26Mb), this includes branchs kisses, turtling on the ground, and some propeller tree cuting job
[BTFL_cli_XbeeLB1_LR_6P_20191017_171746.txt](https://github.com/betaflight/bidircfg/files/3743816/BTFL_cli_XbeeLB1_LR_6P_20191017_171746.txt)

maGfpv, 2019-10-18:
I got the same behavior some days ago. But it happens always when you think its gone. And than I created the issue

benedikt-bartscher, 2019-10-19:
I have the same problem, always getting some DShot erros on Motor 1 no matter what config!

nerocide, 2019-10-21:
Problem happened yesterday on 1 flight, and i was able to capture log, i'll upload it later today.
After the crash caused by the deathroll, i noticed that only one motor would arm, i guess it was motor 3.

maGfpv, 2019-10-23:
Another user reported on YouTube that Bidir/rpm filter is working for him with bf 4.1. on h7extreme
Interesting, he had issues with  motor signals on Motor 3 and 4 with errors above 50 % when he installed a buzzer to the board. Without buzzer everything seems fine. 

I have also installed LEDs and a Buzzer. There were Users reporting that the 5v line of the board is maybe not working correctly @hydra?

@nerocide please upload the log. that would be great. 

@nerocide @benedikt-bartscher @haslinghuis @Mateyhv have you guys installed something that is using the 5v line of the fc?

benedikt-bartscher, 2019-10-23:
Yes, i have installed a buzzer, my receiver and gps to the 5V line!

hydra, 2019-10-23:
@maGfpv it's doubtful there's anything wrong with the 5v supply, if there was the two 3v3 regulators which takes their power from the 5v line would have issues and the CPU would probably not even boot.

As part of the QA process for the H7EXTREME the voltage from the 5v supply must be within a certain range and this range is very tight indeed.  Keeping the range very tight immediately shows up instances up when other devices on the 5v rail are bad, e.g. RSSI circuit, buzzer and transponder circuits as the voltage drops below the QA threshold.  I also do random inspections of the 3 power lines for each batch on the scope to verify there's no unusual noise or output ripple from the 5V and 3V regulators.  Every component on the board is tested, using a real buzzer and real transponder LED and so on.

If you or anyone has access to a scope please send me scope traces for investigation - use the shop contact form for this https://shop.seriouslypro.com/contact/

Additionally, the capacitors for the 3v3 input and output are actually **4x over-specified** for the regulators to make sure there's always power available and that it is clean as possible.

haslinghuis, 2019-10-23:
Ok. Maybe this helps. My ESC init is not out of tune anymore. From the wiki (https://github.com/betaflight/betaflight/wiki/DSHOT-ESC-Protocol):

Originally Posted by airmaxx23

I recently finished building a new quad with the BFF3 controller and Armattan 20a ESCs with d-shot 300 enabled. The first thing I've noticed is that the ESCs tones are out of sync when I plug a battery in and I'm having a flip/roll of death once in a while. What should I check first?

Answer from AILERON8

I've been reloading the BLHeli configurator and reflashing the ESCs, then setting the ESCs to MS, syncing my FC/ESC loop, and calibrating. Then I've been going back to DShot and setting things up normally. Don't ask me why or how it works, but I've had to do this on more than one occasion now or I get a distant ESC song or two playing out of sequence in the background like you have now. This worked each time, but not saying it will work for you with certainty.

haslinghuis, 2019-10-23:
So to do this:
1. Remove lipo, attach FC to PC and open BF configurator and setup Multishot on config tab and Save and reboot.
2. Now reconnect and go to motor tab. Props off (in case off) and enable understand the risks. Now set the master slider to full throttle (2000). Now plugin The LiPo. Enjoy the music. After finished set the master slider back to zero. Enjoy again the music. ESC Calibration is done. I disconnected everything and reattached and reconnected. In BF now set the protocol back to DSHOT (I use 300). Save and Reboot. Reconnect. Enable Bidirectional DSHOT and set your motor_poles if necessary. Save and Reboot and please test (fly) again.

maGfpv, 2019-10-24:
Interesting, I will try this and report.